PlotJuggler/PlotJuggler #1071: Unable to install plotjuggler

ajoardar007, 2025-05-04:
Hello,
I have been unable to install plot-juggler on my machine.

OS: Ubuntu 22.04

Attempted Snap. Error:
`Desktop file didn't specify the Exec field ` on drop-down pop-up.

Attempted `sudo apt install ros-$ROS_DISTRO-plotjuggler-ros -f`. Output:
```
sudo aptitude install ros-$ROS_DISTRO-plotjuggler-ros
The following NEW packages will be installed:
  binutils-dev{ab} libqt5svg5-dev{a} libqt5websockets5{a} 
  libqt5websockets5-dev{a} libqt5x11extras5-dev{a} 
  ros-noetic-plotjuggler{a} ros-noetic-plotjuggler-ros 
  ros-noetic-ros-type-introspection{a} 
0 packages upgraded, 8 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.0 MB of archives. After unpacking 56.8 MB will be used.
The following packages have unmet dependencies:
 binutils-dev : Depends: binutils (= 2.34-6ubuntu1) but 2.34-6ubuntu1.9 is installed
                Depends: libbinutils (= 2.34-6ubuntu1) but 2.34-6ubuntu1.9 is installed
                Depends: libctf0 (= 2.34-6ubuntu1) but 2.34-6ubuntu1.9 is installed
                Depends: libctf-nobfd0 (= 2.34-6ubuntu1) but 2.34-6ubuntu1.9 is installed
The following actions will resolve these dependencies:

     Keep the following packages at their current version:
1)     binutils-dev [Not Installed]                       
2)     ros-noetic-plotjuggler [Not Installed]             
3)     ros-noetic-plotjuggler-ros [Not Installed]         

Accept this solution? [Y/n/q/?]  Y 
```
The installed dependencies version-number exceed the required version number. Maybe this is the problem?

anirudlappathi, 2025-05-05:
<html><head></head><body><p>Yes, you're absolutely right: the error you’re encountering is due to <strong>version conflicts</strong> with the <code inline="">binutils</code> family of packages.</p>
<p>Here's a clear breakdown and a few <strong>recommended solutions</strong> to install <strong>PlotJuggler</strong> on Ubuntu 22.04:</p>
<hr>
<h3>🔍 <strong>Problem Summary</strong></h3>
<p><code inline="">binutils-dev</code> required by <code inline="">ros-noetic-plotjuggler-ros</code> is <strong>strictly</strong> asking for version <code inline="">2.34-6ubuntu1</code>, but your system has <code inline="">2.34-6ubuntu1.9</code>.</p>
<p>This happens because:</p>
<ul>
<li>
<p>ROS Noetic was designed for <strong>Ubuntu 20.04</strong>, not 22.04.</p>
</li>
<li>
<p>Ubuntu 22.04 may have slightly newer versions of the system libraries (like <code inline="">binutils</code>).</p>
</li>
</ul>
<hr>
<h3>✅ <strong>Recommended Solutions</strong></h3>
<h4>✅ Option 1: Use ROS 2 or a Container (Docker)</h4>
<p>If possible, consider using:</p>
<ul>
<li>
<p><strong>ROS 2</strong> (e.g., Foxy or Humble), which is better supported on Ubuntu 22.04.</p>
</li>
<li>
<p>Or install PlotJuggler in a <strong>Docker container</strong> with Ubuntu 20.04 + ROS Noetic.</p>
</li>
</ul>
<pre><code class="language-Dockerfile">FROM ros:noetic-ros-core

RUN apt update &amp;&amp; apt install -y \
  ros-noetic-plotjuggler-ros \
  ros-noetic-rqt \
  &amp;&amp; apt clean
</code></pre>
<p>This avoids host package conflicts.</p>
<hr>
<h4>✅ Option 2: Build PlotJuggler from Source (Recommended for ROS Noetic on Ubuntu 22.04)</h4>
<p>This avoids package version mismatches.</p>
<h5>🔧 Steps:</h5>
<ol>
<li>
<p><strong>Install dependencies:</strong></p>
</li>
</ol>
<pre><code class="language-bash">sudo apt update
sudo apt install qtbase5-dev qttools5-dev libqt5svg5-dev \
                 libqt5websockets5-dev libqt5x11extras5-dev \
                 libboost-system-dev libboost-filesystem-dev \
                 libtinyxml2-dev
</code></pre>
<ol start="2">
<li>
<p><strong>Clone PlotJuggler &amp; ROS plugin:</strong></p>
</li>
</ol>
<pre><code class="language-bash">mkdir -p ~/plotjuggler_ws/src
cd ~/plotjuggler_ws/src

git clone https://github.com/facontidavide/PlotJuggler.git
git clone https://github.com/facontidavide/plotjuggler-ros-plugins.git
</code></pre>
<ol start="3">
<li>
<p><strong>Build the workspace:</strong></p>
</li>
</ol>
<pre><code class="language-bash">cd ~/plotjuggler_ws
cmake -S . -B build
cmake --build build --target install -j$(nproc)
</code></pre>
<ol start="4">
<li>
<p><strong>Run PlotJuggler:</strong></p>
</li>
</ol>
<pre><code class="language-bash">~/plotjuggler_ws/build/PlotJuggler
</code></pre>
<hr>
<h4>✅ Option 3: Use Flatpak (Alternative to Snap)</h4>
<p>Flatpak avoids system-level dependencies like binutils.</p>
<pre><code class="language-bash">sudo apt install flatpak
flatpak install flathub org.plotjuggler.PlotJuggler
flatpak run org.plotjuggler.PlotJuggler
</code></pre>
<hr>
<h3>🧼 (Optional) Clean Up Snap Installation</h3>
<p>If <code inline="">snap</code> is causing issues (e.g., desktop file popup):</p>
<pre><code class="language-bash">sudo snap remove plotjuggler
sudo rm ~/.local/share/applications/snap.plotjuggler*
</code></pre>
<hr>
<h3>Summary</h3>

Solution | Recommended for | Difficulty | Notes
-- | -- | -- | --
✅ Docker | Isolated ROS setup | Medium | No version issues
✅ Build from Source | You want ROS Noetic | Medium | Most flexible
✅ Flatpak | Just need GUI logs | Easy | Not ROS-integrated
❌ apt on 22.04 for Noetic | Not compatible | Not ideal | Version mismatch

facontidavide, 2025-12-07:
i am closing this because it is ROS1, that I don't actively support anymore, and because the Ai generated "reply" is insulting