PlotJuggler/PlotJuggler #1072: Fine grained line customisation and checkboxes

mcamurri, 2025-05-06:
I'm a long time user of `rqt_multiplot` (https://github.com/ANYbotics/rqt_multiplot_plugin) and `signal-scope` (https://github.com/ori-drs/signal_scope)

The former has almost all the features I wanted from a plotting tool, but crashes so often that has become unusable. 

The latter is simple but doesn't have all of the features I need. 

Then I came across Plotjuggler.

Plotjuggler is awesome and has more feature than I dreamed of. 
I love the ability to load from CSV files and the tabs in particular.

However, it lacks a couple of features the above mentioned programs have:

1) rqt_multiplot supports individual lines customisation, in particular each line can be modified in thickness, type, and can have anti aliasing
[image]

2) signal-scope has checkboxes so a signal can be temporarely hidden:

[image]

I know it is a lot to ask, but I was wondering if any of these features are being considered or have been implemented already without me realizing.

thomueller, 2025-05-24:
You can hide a curve by clicking on the label in the legend.
For the line width I am also looking for a solution, because the standard width is nearly invisible on my computer.

zdavkeos, 2025-06-13:
Yes, https://github.com/zdavkeos/PlotJuggler/commit/fffafe7cac8881ee569752b5e8298ff5ccc23d78 allows you to change the line thickness. I don’t know if @facontidavide  doesn’t like it, or if he’s just been busy.

thomueller, 2025-06-14:
I think it looks is really nice, this is basically the only reason, I am looking into compiling it on my own. Thansk for your work on it.