PlotJuggler/PlotJuggler #1074: E: Unable to locate package ros-humble-plotjuggler-ros

tizianofiorenzani, 2025-05-16:
Is the ROS humble package down? It was fine until this morning.

```
sudo apt install ros-humble-plotjuggler-ros
E: Unable to locate package ros-humble-plotjuggler-ros
```

scastro-bdai, 2025-05-16:
it was removed in the latest Humble sync: https://discourse.ros.org/t/new-packages-for-humble-hawksbill-2025-05-16/43801#p-93275-removed-packages-3-4

slyandsmart, 2025-05-26:
Oh, reading this Issue would save me time. 
Don`t forget to add dependencies and all three repos when building it by hand. 

@scastro-bdai do you know will it be added in future again? 
Otherwise i would have to reconsider my workflow.

sandro-meier, 2025-05-30:
> working on it

Great to hear! Do you have a timeline? I'm wondering whether we should build from source or wait for the package to be available again.

Samaksh6398, 2025-06-06:
@slyandsmart 
> Don`t forget to add dependencies and all three repos when building it by hand.
Can you please tag or mention those repos ?

facontidavide, 2025-06-06:
[image]

closed 2025-06-06 by facontidavide