PlotJuggler/PlotJuggler #1075: Build from source error, MacOS 15.4.1, cmake fails from command "cmake --build build/PlotJuggler --config RelWithDebInfo --target install"

memz-life-dev, 2025-05-16:
I use plotjuggler all the time at work on Ubuntu PCs, would love to use on my personal Mac as well but get a cmake failure as described in title.

Complete output:

`(base) andys-MBP:plotjuggler_ws andy$ cmake --build build/PlotJuggler --config RelWithDebInfo --target install
[  0%] Built target plotjuggler_qwt_autogen_timestamp_deps
[  1%] Built target plotjuggler_qwt_autogen
[ 17%] Built target plotjuggler_qwt
[ 17%] Built target lua_autogen_timestamp_deps
[ 17%] Built target lua_autogen
[ 24%] Built target lua
[ 24%] Built target plotjuggler_base_autogen_timestamp_deps
[ 24%] Built target plotjuggler_base_autogen
[ 24%] Building CXX object CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/reactive_function.cpp.o
In file included from /Users/andy/plotjuggler_ws/src/PlotJuggler/plotjuggler_base/src/reactive_function.cpp:7:
In file included from /Users/andy/plotjuggler_ws/src/PlotJuggler/plotjuggler_base/include/PlotJuggler/reactive_function.h:11:
In file included from /Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/sol.hpp:52:
In file included from /Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/stack.hpp:27:
In file included from /Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/trampoline.hpp:27:
In file included from /Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/types.hpp:28:
In file included from /Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/optional.hpp:33:
/Users/andy/plotjuggler_ws/build/PlotJuggler/_deps/sol2-src/include/sol/optional_implementation.hpp:2194:10: error: no member named 'construct' in 'optional<type-parameter-0-0 &>'
 2194 |                         this->construct(std::forward<Args>(args)...);
      |                         ~~~~  ^
1 error generated.
make[2]: *** [CMakeFiles/plotjuggler_base.dir/plotjuggler_base/src/reactive_function.cpp.o] Error 1
make[1]: *** [CMakeFiles/plotjuggler_base.dir/all] Error 2
make: *** [all] Error 2
(base) andys-MBP:plotjuggler_ws andy$ `

Has anyone seen this or is aware of any workarounds?

JarvisDong, 2025-05-19:
Consider replacing "construct" with "emplace".

facontidavide, 2025-05-19:
this is a 3rd party llibrary

oursland, 2025-05-21:
It's been corrected upstream.  This patch will get you past that.

```
diff --git a/cmake/find_or_download_dependencies.cmake b/cmake/find_or_download_dependencies.cmake
index f0708c83..75e5548f 100644
--- a/cmake/find_or_download_dependencies.cmake
+++ b/cmake/find_or_download_dependencies.cmake
@@ -75,7 +75,7 @@ function(find_or_download_dependencies)
 
     ########################################
 
-    CPMAddPackage("gh:ThePhD/sol2@3.3.0")
+    CPMAddPackage("gh:ThePhD/sol2@3.5.0")
 
     ########################################
 
```