PlotJuggler/PlotJuggler #1079: backward-cpp not found when building 3.10.3 on Jazzy

marcoag, 2025-05-22:
It seems that after [moving backward-cpp to `CPMAddPackage`](https://github.com/facontidavide/PlotJuggler/commit/63ced5aa2b20eacd5f90f93ea6f093cbe1dc9e31) the backward-cpp dependency is properly not found when building the package through the buildfarm:
```
//The directory containing a CMake configuration file for backward-cpp.
backward-cpp_DIR:PATH=backward-cpp_DIR-NOTFOUND

[...]

CMake Error at plotjuggler_app/CMakeLists.txt:110 (add_backward):
  Unknown CMake command "add_backward".
```

Full log: https://build.ros2.org/view/Jbin_uN64/job/Jbin_uN64__plotjuggler__ubuntu_noble_amd64__binary/64/console

I tried to give it a shot but I can't reproduce locally and couldn't find a quick fix. Might give it a shot later if I get some more time.

A temporal mitigation would be to go back to `3.9.2` to avoid the regression on the [upcoming sync](*).  @facontidavide if we choose to go that way please advise if additionally we need to rollback to a specific version of `plotjuggler_ros`.

facontidavide, 2025-05-22:
just pushed a fix (hopefully)

closed 2025-05-22 by facontidavide

facontidavide, 2025-05-22:
I will wait to Jenkins to tell me if it is happy. If it is, I will create 3.10.4

marcoag, 2025-05-22:
Thanks for the fast fix!

commit f4098cdb (2025-05-22): add backward again, fix #1079