PlotJuggler/PlotJuggler #1090: PJ cannot be build with modern compilers (clang-17 on Mac) due to sol2-3.3.0

ibrhm0v, 2025-06-11:
I tried to compile PlotJuggler (version 3.10.6) from source on my Mac with these properties:
 
- M4 Pro
- MacOS 15.3.1
- clang-17.0.0

After following [COMPILE.md](https://github.com/facontidavide/PlotJuggler/blob/3.10.6/COMPILE.md) instructions, I have been getting this error:

`/.../plotjuggler_ws/src/PlotJuggler/3rdparty/sol2-3.3.0/include/sol/sol.hpp:6755:10: error: no member named 'construct' in 'optional<type-parameter-0-0 &>'
 6755 |                         this->construct(std::forward<Args>(args)...);`

After digging into the problem. I found out that sol2-3.3.0 is making illegal usage of `optional` by trying to construct an object inside an `optional<T&>`

I fixed this by cloning updated version of sol2 and editing cmake files and include directives accordingly.  I can make a fork and raise a PR upon review of this issue.

ibrhm0v, 2025-06-12:
Also there is a problem with protobuf version. Seems like recent versions of `google/protobuf/compiler/importer.h` started using absl:string_view instead of std::string& and there is a change in functions of `FileErrorCollector` class (AddError became RecordError and AddWarning became RecordWarning)

I will make a fork and commit these changes an raise a PR ASAP

zdavkeos, 2025-06-14:
Updating to Sol2 v3.5.0 allowed me to compile as well.

facontidavide, 2025-06-14:
I can do that upgrade...

ibrhm0v, 2025-07-01:
Ok now I saw that you already did that one @facontidavide

ibrhm0v, 2025-07-01:
Reopened as fix for support for new protobuf compilers PR (#1106 ) is ongoing

closed 2025-07-07 by facontidavide