PlotJuggler/PlotJuggler #1114: bug: scroll step size

dakejahl, 2025-07-10:
Scroll step size is too big even when step size is set to 0. For a given tab with possibly multiple plots, it would be nice if the step scroll was automatically set to the smallest interval of the currently plotted topics. This would make it easier to analyze timing.

For example this is 50Hz data

https://github.com/user-attachments/assets/eca7ada8-38ee-4aaf-855e-c4e0d89312b4

facontidavide, 2026-10-01:
Checking this report uncovered a separate PJ4 port regression: changing the nonzero Step value only stored an unused engine value, so arrow keys ignored it. [PR #1443](https://github.com/PlotJuggler/PlotJuggler/pull/1443) restores the configured step, reapplies it as the data range grows, and guards empty/tiny ranges and non-finite input. The actual keyboard regression and seven related playback/timeline tests pass.

The requested automatic smallest plotted-topic sampling interval when Step is zero is still a feature request, so this issue remains open. Added PJ4 alongside PJ3 because the control regression is present on `main-4.x`.

facontidavide, 2026-11-04:
Sorry that we haven't been able to resolve this report in PJ3. We no longer support or actively maintain PlotJuggler 3; development will focus exclusively on PlotJuggler 4 (PJ4).

We still accept community pull requests for PJ3, although reviewing and merging them is not a priority. We are closing this PJ3 bug report as not planned.

The separate PJ4 regression where a nonzero Step value was ignored is fixed by merged PR #1443. Automatic selection of the smallest plotted-topic sampling interval when Step is zero remains a feature request; it is not being marked implemented by this closure. If that automatic-step capability is needed in PJ4, this issue can also be reopened and tracked as a PJ4 feature request.

If the same or a similar problem occurs in PJ4, please reopen this issue with the PJ4 version, operating system and steps to reproduce it, plus a small example recording or relevant logs where possible. Thank you for reporting it.

closed 2026-11-04 by facontidavide