PlotJuggler/PlotJuggler #1133: Error in opening  bag files since update to latest version

Imti-git, 2025-07-28:
Ever since it updated to version 3.10.11 no bag file can be opened. It shows the following error:

[image]

Imti-git, 2025-07-29:
Hey man, sure...here's the zip file where you can find couple of bag files. I tried to upload all of them.

[field test files.zip](https://github.com/user-attachments/files/21488540/field.test.files.zip)

YangLingyuan, 2025-07-30:
Run into the same issue here.  Loading all the previously working bags fail after updated to 3.10.11.
The terminal prints the following error: "[ERROR] [1753844354.179070330] [rosbag2_storage]: Could not load/open plugin with storage id 'sqlite3'"

facontidavide, 2026-09-20:
That error came from the rosbag2 'sqlite3' storage plugin not being bundled in the PJ3 snap. PlotJuggler 4 loads ROS bags via MCAP (the modern rosbag2 default) and doesn't use the sqlite3 storage path at all, so this packaging failure no longer applies.

---
*PlotJuggler 4 is now the official release and the default branch of this repo; PlotJuggler 3 is end-of-life and no longer receives features or fixes. If this still affects you in PJ4, please reopen or file a fresh issue — it will be looked at.*

closed 2026-09-20 by facontidavide