PlotJuggler/PlotJuggler #1167: MQTT subscriber connect failed

klhappy, 2025-09-17:
1. chossing mqtt 3.1.1 in mqtt configuration
2. clicking the “connect” button
3. a waring is happen，“client disconnected.Maybe a problem with authentication”
but chossing mqtt5 is OK
[image]

BenArtes, 2026-01-14:
Not sure if related or not, but  I had issues connecting and selecting topics until I re-typed the default values. 
Specifically:
 * Hostname default value 'localhost' doesn't work, have to manually type 'localhost' into the field
 * Topics filter default value '#' also doesn't work, have to manually type '#' into the field

Once I typed in the default values, things worked as expected

facontidavide, 2026-10-01:
A small PJ3 repair is ready in https://github.com/PlotJuggler/PlotJuggler/pull/1441: empty/default host and topic filter values are restored correctly, and the initial subscription configuration is assigned before the network callback can use it. Both paths are reproduced and checked, including a retained-message test with Mosquitto 2.0.22. I could not reproduce the original protocol-specific disconnect with explicit settings on Linux, so this issue stays open. PJ4 uses a different MQTT client; its current shared connection helper received data with MQTT 3.1, 3.1.1 and 5 in the same broker test.

facontidavide, 2026-11-04:
Sorry that we haven't been able to resolve this report in PJ3. We no longer support or actively maintain PlotJuggler 3; development will focus exclusively on PlotJuggler 4 (PJ4).

We still accept community pull requests for PJ3, although reviewing and merging them is not a priority. We are closing this PJ3 bug report as not planned.

Merged PR #1441 fixes PJ3's default MQTT settings and an initial subscription race. The original protocol-specific disconnect was not reproduced, so this closure does not claim that symptom is fixed.

If the same or a similar problem occurs in PJ4, please reopen this issue with the PJ4 version, operating system and steps to reproduce it, plus a small example recording or relevant logs where possible. Thank you for reporting it.

closed 2026-11-04 by facontidavide