PlotJuggler/PlotJuggler #1170: Fields missing in left-hand pane after loading a layout

jepemi, 2025-09-19:
## Description

When loading a layout, some fields disappear from the left-hand pane, even though they are still being plotted correctly.

## Steps to Reproduce

1. Start PlotJuggler.
2. Start streaming data, there is an size-three array field `b`
3. Save a layout that plots specific fields (example: `b[1]`).
4. Close and restart PlotJuggler.
5. When prompted: Start the previously used streaming plugin? select `Yes`.
6. Load the previously saved layout.

## Observed Behavior

The field plotted in the layout (e.g., `b[1]`) remains visible in the plot.
However, it disappears from the left-hand Timeseries List pane.
Other related fields (e.g., `b[0]`, `b[2]`) remain visible.

This seems related to (or the same as) #1031 , though reported with a different plugin.

Reproduced both with PlotJuggler 3.9.1 and 3.12.0

[image]

facontidavide, 2025-09-19:
I am checking this: https://github.com/eProsima/fastdds-visualizer-plugin/blob/main/plugins/datastreamer_plugin/datastreamer/FastDdsDataStreamer.cpp

At a first look it seems to be correct. Don't know where the issue might be. I will keep this in mind to see if it can reproduced using other streamers

sastel, 2025-09-19:
@facontidavide might it be related to https://github.com/facontidavide/PlotJuggler/issues/1031?

sastel, 2025-09-19:
Oops never mind I see that @jepemi already referenced it.

facontidavide, 2026-11-04:
Sorry that we haven't been able to resolve this report in PJ3. We no longer support or actively maintain PlotJuggler 3; development will focus exclusively on PlotJuggler 4 (PJ4).

We still accept community pull requests for PJ3, although reviewing and merging them is not a priority. We are closing this PJ3 bug report as not planned.

If the same or a similar problem occurs in PJ4, please reopen this issue with the PJ4 version, operating system and steps to reproduce it, plus a small example recording or relevant logs where possible. Thank you for reporting it.

closed 2026-11-04 by facontidavide